paulvee/100MHz-Differential-Probe #1: Gerber

iskralpust, 2020-04-30:
Hi I see you have removed the Gerber files for this project. Are you adding new files? It would also be super great is you can add the Diptrace-Eagle files. Thanks.

paulvee, 2020-05-01:
Yes I did. I wanted to verify the new layouts before I posted the Gerbers. The boards just arrived and I will be adding the components and start testing. very soon. Keep an eye on my blog for this project.

iskralpust, 2020-05-02:
Awesome looking forward to hearing how it turns out.

iskralpust, 2020-05-17:
Hi is there any chance you can upload the board and schematic files I was going to make some modifications to the board dimensions for an enclosure and add a 9v battery holder and pcb mount power switch.

paulvee, 2020-05-17:
iskralpust,
At this moment we do not want to publish more details, in fear that somebody will copy our design and turn it into a commercial product and sell it without us benefiting. This has happened before as you will probably know.

closed 2020-05-18 by iskralpust