paulvee/100MHz-Differential-Probe #7: Diff Probe BOM V3.xlsx

wocard2, 2024-10-07:
Thanks for the notes in the new BOM file!

About note 1: "resistors" should be replaced with "capacitors"

Note 3 is unreferenced, probably refers to U3. Isn't some minimum output current for U3 required?

About note 4: really exactly equal? Or would a smaller ESR be okay?

Note 6 is unreferenced.

About note 7: the 0402 footprints are extremely large (0402 components barely touch the pads), 0603 components can be easily hand-soldered onto them. Even 0805 tantalum capacitors  could be used for C24 and C31 as those have only one neighbor and suffer the most from their tiny size.

paulvee, 2024-10-08:
Hi wocard2,
Thank you for the inputs, you caught a few more.
I fixed 1, 3, 4, added J2 and deleted 6.
7 is now 6. About size 0402 (1005 Metric), it is listed as 0.043" L x 0.024" W (1.10mm x 0.60mm). There are many 0402 parts on the board, so I think you are looking at the wrong specification for 0402.
I updated the BOM file.

wocard2, 2024-10-08:
Here's the ocular proof (please excuse the poor image quality):
[image]
I placed a 0805 (imperial) resistor on the C24 footprint (yes, it's too wide, but that works if there is only one neighbor) , a 0603 (imperial) resistor on the C25 footprint, and a 0402 (imperial) resistor on the C36 footprint. All these footprints are supposed to be 0402 (imperial) footprints. As you can see, a 0603 or even a 0805 capacitor can be used for C24. The advantage, besides availability, is a much smaller ESR. Those 0402 (imperial) footprints look like 0603 (imperial) footprints to me.

wocard2, 2024-10-08:
Any requirements for the minimum output power of U3?

paulvee, 2024-10-08:
U3 only supplies the common ground, a few mA should do. Look at the LM7301 datasheet for guidance.
I uploaded two pictures of my probe that will show the sizes of the parts I used, which are specified in the BOM.

closed 2024-10-08 by paulvee

wocard2, 2024-10-08:
Thanks for the great images! Whatever equiment I use for taking photoghraphs, the result will be mediocre.

The image of the front side clearly shows that the footprint of C26 is too big (or that the component is too small). That could explain the tombstoning you experienced. That image also shows that your C24 is bigger than 0402. Looks like 0603. You might want to mention in the BOM that 0603 components will fit on those 0402 footprints, in particiular for C24 and C31.

wocard2, 2024-10-21:
Note that the CL05A475MP5NRNC listed for C24 and C31 in the V3a BOM has only 1.1 µF at 5.35 V DC. 0603 MLCCs are better than 0402 MLCCs w.r.t. DC bias...

paulvee, 2024-11-03:
There is a Tantalum Poly available from Digikey : 478-10032-1-ND.
When you use an X5R, use a 10uF version