paulvee/100MHz-Differential-Probe #8: M1

wocard2, 2024-10-10:
I found no version of the BSS84 that has a typical (not to mention a maximum) RDSon below 1Ω (at VGS=-10V, let alone at VGS=-7.5V) as required by note 5. For the BSS84 specimen mentioned in the BOM, RDSon is specified for VGS=-5V only, with a maximum value of 10Ω at 25°C.

An additional requirement for M1 is a maximum VGS of ±12 V or better. That's not mentioned by note 5.

For a maximum VGS of ±20 V or better (to add some margin of safety), choices for P-channel MOSFETS with low RDSon are very limited. It might be better to use an N-channel MOSFET on V- instead of a P-channel MOSFET on V+.

paulvee, 2024-10-11:
DigiKey has more than 60.000 of the BSS84 in stock. LCSC has 5600+ in stock.

wocard2, 2024-10-11:
Perhaps I was not clear enough. There are two issues:
(1) the BSS84 does not meet the requirement you impose on substitutions for the BSS84
(2) there's a requirement (on VGS) missing. For instance, the AO3435 would work according to note 5, but it will blow up.

wocard2, 2024-10-11:
In fact, I used an AO3435 but noticed the problem before applying power.

paulvee, 2024-10-11:
Bud uses an AO3401 in his probe.

wocard2, 2024-10-11:
I'm now also using an AO3401A. It's a bit risky as its absolute maximum value for VGS is ±12V. At least, it has a better RDSon than the BSS84.